spacetelescope/acstools #9: dask?!

pllim, 2016-07-28:
As [reported by Joe here](https://github.com/spacetelescope/acstools/pull/8#issuecomment-235969208)

pllim, 2016-07-28:
@jhunkeler , what if we change `install_requires` to just `requires` in setup?

jhunkeler, 2016-07-28:
Apparently `requires` does not conform to the same parsing conventions as `install_requires`:

```
+ python setup.py install
Traceback (most recent call last):
  File "setup.py", line 58, in <module>
    zip_safe=False
  File "/home/jhunk/anaconda3/envs/_build/lib/python2.7/distutils/core.py", line 111, in setup
    _setup_distribution = dist = klass(attrs)
  File "/home/jhunk/anaconda3/envs/_build/lib/python2.7/site-packages/setuptools-23.0.0-py2.7.egg/setuptools/dist.py", line 272, in __init__
  File "/home/jhunk/anaconda3/envs/_build/lib/python2.7/distutils/dist.py", line 259, in __init__
    getattr(self.metadata, "set_" + key)(val)
  File "/home/jhunk/anaconda3/envs/_build/lib/python2.7/distutils/dist.py", line 1220, in set_requires
    distutils.versionpredicate.VersionPredicate(v)
  File "/home/jhunk/anaconda3/envs/_build/lib/python2.7/distutils/versionpredicate.py", line 113, in __init__
    raise ValueError("expected parenthesized list: %r" % paren)
ValueError: expected parenthesized list: '>=1.0'
+ exit 1
Command failed: /bin/bash -x -e /home/jhunk/anaconda3/conda-bld/work/conda_build.sh
```

jhunkeler, 2016-07-28:
Reference from basic python installation method:
```
$ python setup.py install
running install
running bdist_egg
running egg_info
writing requirements to acstools.egg-info/requires.txt
writing acstools.egg-info/PKG-INFO
writing top-level names to acstools.egg-info/top_level.txt
writing dependency_links to acstools.egg-info/dependency_links.txt
writing entry points to acstools.egg-info/entry_points.txt
reading manifest file 'acstools.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching '*' under directory 'scripts'
no previously-included directories found matching 'build'
no previously-included directories found matching 'doc/build'
warning: no previously-included files matching '*.pyc' found anywhere in distribution
warning: no previously-included files matching '*.o' found anywhere in distribution
writing manifest file 'acstools.egg-info/SOURCES.txt'
installing library code to build/bdist.macosx-10.6-x86_64/egg
running install_lib
running build_py
copying acstools/version.py -> build/lib/acstools
creating build/bdist.macosx-10.6-x86_64/egg
creating build/bdist.macosx-10.6-x86_64/egg/acstools
copying build/lib/acstools/__init__.py -> build/bdist.macosx-10.6-x86_64/egg/acstools
copying build/lib/acstools/acs2d.py -> build/bdist.macosx-10.6-x86_64/egg/acstools
copying build/lib/acstools/acs_destripe.py -> build/bdist.macosx-10.6-x86_64/egg/acstools
copying build/lib/acstools/acs_destripe_plus.py -> build/bdist.macosx-10.6-x86_64/egg/acstools
copying build/lib/acstools/acsccd.py -> build/bdist.macosx-10.6-x86_64/egg/acstools
copying build/lib/acstools/acscte.py -> build/bdist.macosx-10.6-x86_64/egg/acstools
copying build/lib/acstools/acsrej.py -> build/bdist.macosx-10.6-x86_64/egg/acstools
copying build/lib/acstools/acssum.py -> build/bdist.macosx-10.6-x86_64/egg/acstools
copying build/lib/acstools/calacs.py -> build/bdist.macosx-10.6-x86_64/egg/acstools
creating build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/pars/acs2d.cfg -> build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/pars/acs2d.cfgspc -> build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/pars/acs_destripe.cfg -> build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/pars/acs_destripe.cfgspc -> build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/pars/acs_destripe_plus.cfg -> build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/pars/acs_destripe_plus.cfgspc -> build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/pars/acsccd.cfg -> build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/pars/acsccd.cfgspc -> build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/pars/acscte.cfg -> build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/pars/acscte.cfgspc -> build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/pars/acsrej.cfg -> build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/pars/acsrej.cfgspc -> build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/pars/acssum.cfg -> build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/pars/acssum.cfgspc -> build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/pars/calacs.cfg -> build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/pars/calacs.cfgspc -> build/bdist.macosx-10.6-x86_64/egg/acstools/pars
copying build/lib/acstools/satdet.py -> build/bdist.macosx-10.6-x86_64/egg/acstools
copying build/lib/acstools/version.py -> build/bdist.macosx-10.6-x86_64/egg/acstools
byte-compiling build/bdist.macosx-10.6-x86_64/egg/acstools/__init__.py to __init__.pyc
byte-compiling build/bdist.macosx-10.6-x86_64/egg/acstools/acs2d.py to acs2d.pyc
byte-compiling build/bdist.macosx-10.6-x86_64/egg/acstools/acs_destripe.py to acs_destripe.pyc
byte-compiling build/bdist.macosx-10.6-x86_64/egg/acstools/acs_destripe_plus.py to acs_destripe_plus.pyc
byte-compiling build/bdist.macosx-10.6-x86_64/egg/acstools/acsccd.py to acsccd.pyc
byte-compiling build/bdist.macosx-10.6-x86_64/egg/acstools/acscte.py to acscte.pyc
byte-compiling build/bdist.macosx-10.6-x86_64/egg/acstools/acsrej.py to acsrej.pyc
byte-compiling build/bdist.macosx-10.6-x86_64/egg/acstools/acssum.py to acssum.pyc
byte-compiling build/bdist.macosx-10.6-x86_64/egg/acstools/calacs.py to calacs.pyc
byte-compiling build/bdist.macosx-10.6-x86_64/egg/acstools/satdet.py to satdet.pyc
byte-compiling build/bdist.macosx-10.6-x86_64/egg/acstools/version.py to version.pyc
creating build/bdist.macosx-10.6-x86_64/egg/EGG-INFO
copying acstools.egg-info/PKG-INFO -> build/bdist.macosx-10.6-x86_64/egg/EGG-INFO
copying acstools.egg-info/SOURCES.txt -> build/bdist.macosx-10.6-x86_64/egg/EGG-INFO
copying acstools.egg-info/dependency_links.txt -> build/bdist.macosx-10.6-x86_64/egg/EGG-INFO
copying acstools.egg-info/entry_points.txt -> build/bdist.macosx-10.6-x86_64/egg/EGG-INFO
copying acstools.egg-info/not-zip-safe -> build/bdist.macosx-10.6-x86_64/egg/EGG-INFO
[... 97 lines cut ...]
Adding d2to1 0.2.12.post1 to easy-install.pth file

Using /Users/jhunk/anaconda3/envs/ugh2/lib/python2.7/site-packages
Searching for Pillow==3.3.0
Best match: Pillow 3.3.0
Adding Pillow 3.3.0 to easy-install.pth file

Using /Users/jhunk/anaconda3/envs/ugh2/lib/python2.7/site-packages
[... cut ...]
Adding networkx 1.11 to easy-install.pth file

Using /Users/jhunk/anaconda3/envs/ugh2/lib/python2.7/site-packages
Searching for six==1.10.0
Best match: six 1.10.0
Adding six 1.10.0 to easy-install.pth file

Using /Users/jhunk/anaconda3/envs/ugh2/lib/python2.7/site-packages
Searching for decorator==4.0.9
Best match: decorator 4.0.9
Adding decorator 4.0.9 to easy-install.pth file

Using /Users/jhunk/anaconda3/envs/ugh2/lib/python2.7/site-packages
Finished processing dependencies for acstools==2.0.2.dev9
```

jhunkeler, 2016-07-28:
```
scikit-image 0.12.3 np111py35_1
-------------------------------
file name   : scikit-image-0.12.3-np111py35_1.tar.bz2
name        : scikit-image
version     : 0.12.3
build number: 1
build string: np111py35_1
channel     : defaults
size        : 17.8 MB
date        : 2016-05-31
fn          : scikit-image-0.12.3-np111py35_1.tar.bz2
license     : 3-clause BSD
license_family: BSD
md5         : b04b674e24288b26839ef00812c23351
priority    : 2
schannel    : defaults
url         : https://repo.continuum.io/pkgs/free/osx-64/scikit-image-0.12.3-np111py35_1.tar.bz2
dependencies:
    matplotlib >=1.3.1
    networkx >=1.8
    numpy 1.11*
    pillow >=2.1
    python 3.5*
    scipy
    six >=1.7.3
```

```
dask 0.10.1 py35_0
------------------
file name   : dask-0.10.1-py35_0.tar.bz2
name        : dask
version     : 0.10.1
build number: 0
build string: py35_0
channel     : defaults
size        : 586 KB
date        : 2016-07-13
fn          : dask-0.10.1-py35_0.tar.bz2
license     : BSD
md5         : 3568c7d96d50a2296f7fe51e62c572fa
priority    : 2
schannel    : defaults
url         : https://repo.continuum.io/pkgs/free/osx-64/dask-0.10.1-py35_0.tar.bz2
dependencies:
    bokeh
    chest
    cloudpickle >=0.2.1
    numpy
    pandas
    partd >=0.3.3
    python 3.5*
    toolz >=0.7.2
```

pllim, 2016-07-28:
Ugh... that's ugly... I'm making it optional. Will file a PR soon. Thanks for investigating!

PR #10 (2016-07-28): Make some dependencies optional
Fixes #9.

@jhunkeler , looks much cleaner now:
```
.../lib/python2.7/site-packages/setuptools-23.0.0-py2.7.egg/setuptools/dist.py:294:
UserWarning: The version specified ('714bc73b') is an invalid version,
this may not work as expected with newer versions of setuptools, pip, and PyPI.
Please see PEP 440 for more details.
running install
running bdist_egg
running egg_info
creating acstools.egg-info
writing requirements to acstools.egg-info/requires.txt
writing acstools.egg-info/PKG-INFO
writing top-level names to acstools.egg-info/top_level.txt
writing dependenc [... cut]
closed 2016-07-28 by pllim