spacetelescope/acstools #12: CTE forward modeler

pllim, 2016-09-07:
There was a CTE forward modeler in `PixCteCorr` (a hidden function), which was removed along with the standalone CTE removal module that we agreed to no longer support. A proposal was raised to bring just the forward modeler back for general use.

pllim, 2016-11-02:
@jamienoss , if you come across such a functionality in the new CTE algorithm, please consider enabling it as a standalone executable in ACSTOOLS.

OlyDLG, 2017-03-26:
Hi!  I'm interviewing later this week for a position under Warren Hack, so I did a little research, identified acstools as a component on which he is "Supervising Developer" and found this sole open issue, so I thought I'd check it out.  Might I ask: in this context, what does CTE stand for?  Thanks!

pllim, 2017-07-11:
@jamienoss , any chance there is a forward modeler somewhere in the code you ported from Jay for 2017.2a?

pllim, 2018-06-01:
It is implemented over at spacetelescope/hstcal#313

closed 2018-06-01 by pllim