spacetelescope/acstools #25: Update CALACS doc to include new Gen 2 options

pllim, 2017-06-28:
This needs to be updated:
http://acstools.readthedocs.io/en/latest/calacs_hstcal.html

Source is here:
https://github.com/spacetelescope/acstools/blob/master/doc/source/calacs_hstcal.rst

Also, did this change?
http://acstools.readthedocs.io/en/latest/calacs.html

jamienoss, 2017-06-28:
None of the command line options were propagated through to ``ACSTOOLS`` so http://acstools.readthedocs.io/en/latest/calacs.html does not need to be changed.

jamienoss, 2017-06-28:
Thanks for pointing this out BTW :)

jamienoss, 2017-06-30:
How or when etc is http://acstools.readthedocs.io/en/latest/calacs_hstcal.html built/updated? is this done manually or...?

jamienoss, 2017-06-30:
Is there an ISR for the new CTE correction? (I'll email Norman)

pllim, 2017-06-30:
ReadTheDocs builds the doc whenever you commit to `master`.

jamienoss, 2017-06-30:
The docs shouldn't be built and updated from master. They should be building the last release only, right?

pllim, 2017-06-30:
ReadTheDocs builds from `master` and release tags. The link you point out is "latest", thus points to `master`. If you replace "latest" with "stable", it should point to latest release.

jamienoss, 2017-06-30:
Awesome, thanks.

PR #27 (2017-06-30): Update CALACS doc to include new CTE Gen 2 options
resolves #25
commit d9fcd8a7 (2017-06-30): Update CALACS doc to include new CTE Gen 2 options

resolves #25
closed 2017-06-30 by jamienoss