spacetelescope/acstools #26: Add gen2 CTE ISR to docs

jamienoss, 2017-06-30:
At the time of resolving #25 no new ISR was available. When this is available, it needs to be integrated into the docs.

jamienoss, 2017-06-30:
@JayVB Norman stated that you are authoring this? If you could keep me posted on when this is complete, that would be great, thanks.

jamienoss, 2017-06-30:
You mean the algorithm didn't change much from that used in calwf3, however, the change from the correction algorithm used in calacs is significant. Isn't this ISR for ACS or is it something different? I'm completely unfamiliar with the nature of an ISR.

JayVB, 2017-06-30:
I'm putting together an ISR (Instrument Science Report) on the pinning of the model in ACS/WFC and WFC3/UVIS... it's a joint report, but Norma and Elena may prefer to have it separated.  I'll talk to them in July.  Yes, the ACS algorithm is different from before, but it's not that different.... the fundamental idea is the same.

closed 2019-02-27 by pllim