spacetelescope/acstools #28: Fix satdet regression test failure

pllim, 2017-07-11:
Satellite detection trail regression test in ACSTOOLS started failing on June 14, 2017. At first, I thought it was due to astropy/astropy#6207 but that was not merged into Astropy dev until June 15. Both code and test data have not been recently changed, so it is likely that some behavior change upstream (scipy?) caused this.

https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=102860998

c/c @dborncamp

dborncamp, 2017-07-11:
I see the test failing in the pandokia engine but can I get at the data somehow?

dborncamp, 2017-07-11:
I cannot reproduce this error. Could you give me more information on the environment?

pllim, 2017-07-11:
@jhunkeler or @rendinam know where to dig that out.

pllim, 2017-07-11:
@dborncamp , by the way, in the test log given above, don't worry about the traceback at the end, but rather note that `fitsdiff` reported differences in result, which I think is also what you see?

pllim, 2017-07-11:
Update: Dave Borncamp was unable to produce the bad output on his environment but I am able to... My bet is on some changes upstream between Astropy 1.3 and 2.0. Maybe astropy/astropy#5991 but why it did not show up until June 14 on dev (which is supposed to pick up Astropy dev), I don't know.

pllim, 2017-07-13:
There is still some differences but they appear to me as aliasing effect: https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=105008197

@dborncamp, those files are temporarily at `/eng/ssb/lim/acstools_issue29`. Please have a look and see if the differences are acceptable. Thanks!

dborncamp, 2017-07-17:
These changes are good. Thanks @pllim!

PR #30 (2017-07-12): Astropy 2.0 compat for satdet
Fix #28 . Also PEP8 clean-up.

Looks good to you, @dborncamp? I can't add you as a formal reviewer but I would like you to look at it before merge. Thanks!
closed 2017-07-17 by pllim