spacetelescope/acstools #31: ACSCTE Python wrapper cannot run Gen 1

pllim, 2017-07-21:
There is no option to pass in `--ctegen 1`, causing this test failure:

https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=106065914

```
PCTECORR PERFORM
PCTEFILE pctetab_cte.fits
Trying to open pctetab_cte.fits...
Read in Primary header from pctetab_cte.fits...
(pctecorr) Generation 1 PCTETAB file auto-detected.
ERROR:    (pctecorr) Gen1 algorithm detected in PCTETAB (CTE_NAME). Default is g
en2, use '--ctegen 1' to override.
ERROR:    Error processing jbny01syq_blv_tmp.fits.
ERROR:    Error processing jbny01syq_blv_tmp.fits.
ERROR:    Calibration file(s) missing.
```

jamienoss, 2017-07-21:
Is this an issue from the regression tests or a user?

I didn't implement this in ``acstools`` due to Norman's request for minimal awareness. This being said since it gives the help message perhaps it doesn't make too much sense to not to, we'll see.

Really ``acstools`` should just be able to pass arbitrary options through to its actual executable, this would at least provide an interim work-around in these cases.

pllim, 2017-07-21:
Not publicizing does not mean not letting people to actually use it, so I prefer adding an option for Gen 1 anyway.

jamienoss, 2017-07-21:
I know, I just meant that if it gets added to ``argparser`` that it'll show up in the help usage and that I'll have to document it too.

I can easily add it. I'll open another issue for the argument back door also.

jamienoss, 2017-10-03:
Solved by #33

closed 2017-10-03 by jamienoss