spacetelescope/acstools #36: ACSCTE: Populating constant-array extension in output file.

nmiles2718, 2017-09-14:
I have noticed that when running ACSCTE on an image with a constant-array extension that the file output by ACSCTE is missing the data from that extension. In this context, `jd4zcob5q_flt.fits` has been processed with BIASCORR and BLEVCORR.
```python
>>> with fits.open('jd4zcob5q_flt.fits',mode='update') as hdu:
...:    # Print hdu info
...:    hdu.info()
...:    # Show hdu[3].data is nonzero
...:    dq2 = hdu[3].data
...:    print(dq2.shape, len(dq2[dq2!=0]))
...:    # Now set extension to be all zeros
...:    hdu[3].data = np.zeros((hdu[3].header['NAXIS2'], hdu[3].header['NAXIS1']))
...:    # Set the PCTECORR keyword to PERFORM
...:    hdu[0].header['PCTECORR'] = 'PERFORM'
...:    # Set PCTETAB to newest version in jref   
...:    hdu[0].header['PCTETAB']='jref$16k1747tj_cte.fits'

Filename: jd4zcob5q_flt.fits
No.    Name         Type      Cards   Dimensions   Format
  0  PRIMARY     PrimaryHDU     252   ()      
  1  SCI         ImageHDU        95   (4096, 2048)   float32   
  2  ERR         ImageHDU        48   (4096, 2048)   float32   
  3  DQ          ImageHDU        40   (4096, 2048)   int16   
  4  SCI         ImageHDU        95   (4096, 2048)   float32   
  5  ERR         ImageHDU        48   (4096, 2048)   float32   
  6  DQ          ImageHDU        40   (4096, 2048)   int16   
(2048, 4096) 136340
>>> acstools.acscte.acscte('jd4zcob5q_flt.fits')

Setting max threads to 8 out of 8 available
Trying to open jd4zcob5q_flt.fits...
Read in Primary header from jd4zcob5q_flt.fits...

CALACSBEG*** ACSCTE -- Version 9.2.0 (01-Jun-2017) ***
Begin    14-Sep-2017 13:15:17 EDT
Input    jd4zcob5q_flt.fits
Output   jd4zcob5q_flt_blc_tmp.fits
Trying to open jd4zcob5q_flt.fits...
Read in Primary header from jd4zcob5q_flt.fits...
APERTURE WFC
FILTER1 F502N
FILTER2 F660N
DETECTOR WFC
DESCRIP read in as: 

DESCRIP read in as: 

DESCRIP read in as: 

CCDTAB   jref$xa81715gj_ccd.fits
CCDTAB   PEDIGREE=inflight
CCDTAB   DESCRIP =July 2009

PCTECORR PERFORM
PCTEFILE jref$16k1747tj_cte.fits
Trying to open jref$16k1747tj_cte.fits...
Read in Primary header from jref$16k1747tj_cte.fits...
(pctecorr) Generation 2 PCTETAB file auto-detected.
(pctecorr) Reading CTE parameters from PCTETAB file: 'jref$16k1747tj_cte.fits'...
Trying to open jref$16k1747tj_cte.fits...
Read in Primary header from jref$16k1747tj_cte.fits...

CTE_NAME: PixelCTE 2017
CTE_VER: 2.0
CTEDATE0: 52334.9
CTEDATE1: 57710.4
PCTETLEN: 60
PCTERNOI: 3.250000
PCTERNFOR: 5
PCTERNPAR: 7
PCTENSMD: 0
PCTETRSH: -10
FIXROCR: 1
Reading in image from extension 3
Reading in image from extension 4
ERROR:    Keyword = `PCTENSMD'.
ERROR:    Keyword = `PCTETLEN'.
ERROR:    Keyword = `PCTERNOI'.
ERROR:    Keyword = `PCTENFOR'.
ERROR:    Keyword = `PCTENPAR'.
ERROR:    Keyword = `FIXROCR'.
(pctecorr) Read noise level PCTERNCL: 3.250000
(pctecorr) Readout simulation forward modeling iterations PCTENFOR: 5
(pctecorr) Number of iterations used in the parallel transfer PCTENPAR: 7
(pctecorr) CTE_FRAC: 1.052591
(pctecorr) PCTETAB read
(pctecorr) Using parallel processing provided by OpenMP inside CTE routine
(pctecorr) Performing CTE correction for amp C
(pctecorr) Calculating smooth readnoise image...
(pctecorr) Time taken to smooth image: 1.76(s) with 8 threads
(pctecorr) ...complete.
(pctecorr) Creating charge trap image...
(pctecorr) Time taken to populate pixel trap map image: 0.00(s) with 8 threads
(pctecorr) ...complete.
(pctecorr) Running correction algorithm...
(pctecorr) ...complete.
(pctecorr) Total count difference (corrected-raw) incurred from correction: 7768892.500000 (4.444590%)
(pctecorr) Performing CTE correction for amp D
(pctecorr) Calculating smooth readnoise image...
(pctecorr) Time taken to smooth image: 2.32(s) with 8 threads
(pctecorr) ...complete.
(pctecorr) Creating charge trap image...
(pctecorr) Time taken to populate pixel trap map image: 0.00(s) with 8 threads
(pctecorr) ...complete.
(pctecorr) Running correction algorithm...
       
(pctecorr) ...complete.
(pctecorr) Total count difference (corrected-raw) incurred from correction: 7965455.500000 (4.196546%)
(pctecorr) CTE run time for current chip: 1891.15(s) with 8 procs/threads

(pctecorr) Using parallel processing provided by OpenMP inside CTE routine
(pctecorr) Performing CTE correction for amp A
(pctecorr) Calculating smooth readnoise image...
(pctecorr) Time taken to smooth image: 2.58(s) with 8 threads
(pctecorr) ...complete.
(pctecorr) Creating charge trap image...
(pctecorr) Time taken to populate pixel trap map image: 0.00(s) with 8 threads
(pctecorr) ...complete.
(pctecorr) Running correction algorithm...
(pctecorr) ...complete.
(pctecorr) Total count difference (corrected-raw) incurred from correction: 8093220.000000 (4.665939%)
(pctecorr) Performing CTE correction for amp B
(pctecorr) Calculating smooth readnoise image...
(pctecorr) Time taken to smooth image: 2.45(s) with 8 threads
(pctecorr) ...complete.
(pctecorr) Creating charge trap image...
(pctecorr) Time taken to populate pixel trap map image: 0.00(s) with 8 threads
(pctecorr) ...complete.
(pctecorr) Running correction algorithm...
(pctecorr) ...complete.
(pctecorr) Total count difference (corrected-raw) incurred from correction: 7466168.500000 (3.768336%)
(pctecorr) CTE run time for current chip: 1897.81(s) with 8 procs/threads

PCTECORR COMPLETE

End      14-Sep-2017 14:22:38 EDT

*** ACSCTE complete ***
>>>with fits.open('jd4zcob5q_flt_blc_tmp.fits') as hdu:
    ...:     hdu.info()
    ...:     print(type(hdu[3].data))
    ...:       
Filename: jd4zcob5q_flt_blc_tmp.fits
No.    Name         Type      Cards   Dimensions   Format
  0  PRIMARY     PrimaryHDU     269   ()      
  1  SCI         ImageHDU        95   (4096, 2048)   float32   
  2  ERR         ImageHDU        48   (4096, 2048)   float32   
  3  DQ          ImageHDU        41   ()      
  4  SCI         ImageHDU        95   (4096, 2048)   float32   
  5  ERR         ImageHDU        48   (4096, 2048)   float32   
  6  DQ          ImageHDU        40   (4096, 2048)   int16   
<class 'NoneType'>

```
Note tha in the output image, the extension I set to be a constant-array with all zeros is now type `NONE` instead of an (4096, 2048) array of all zeros. I would upload an example raw file to use, but for my case the file size is too large as it is as full frame image.

philhodge, 2017-09-14:
This is an HST convention to reduce disk space.  When you open the file with astropy.io.fits as you did in this example, you see a missing data block for the constant extension, but if you opened it using the hstio functions (as is done by the calxxx routines), you would get an array of the correct size and value.  There are keywords in the extension (the third extension, in your example) that give the size and constant value, and the hstio functions create an array and fill it with that value.  astropy.io.fits doesn't follow that convention because it's so HST-specific.

pllim, 2017-09-14:
Yes, this is basically the same as astropy/astropy#6401 (and there is a Python workaround given at https://github.com/astropy/astropy/issues/6401#issuecomment-319130317 ). In this case, you will have to use the workaround until there is a solution for spacetelescope/stak#103 . Sorry for the inconvenience!

closed 2017-09-14 by pllim