spacetelescope/acstools #40: CTE Corrected Associations

tddesjardins, 2017-10-04:
Need to have some kind of logic for what to do if images in an association have different values of PCTECORR. This may happen if, as Norman pointed out, a mix of full-frame and subarray images have been taken of the same scene or if a user has supplied their own custom made association file as we show them how to do in the ACS Data Handbook.

pllim, 2017-10-04:
Not sure if I understand the problem. Doesn't CALACS already read `PCTECORR` from individual files in the association? Are you saying that CALACS is ignoring `PCTECORR` flag when ASN is given? Do you have a reproducible example?

pllim, 2017-10-04:
Also since this issue is open in ACSTOOLS, which specific ACSTOOLS task are you referring to?

tddesjardins, 2017-10-04:
Sorry, should have specified that this affects `acscte` and this is continuing an email that @jamienoss sent around asking what should happen in such a case.

jamienoss, 2017-10-04:
Moving this to HSTCAL https://github.com/spacetelescope/hstcal/issues/242
@tddesjardins acstools is just a set of wrappers for ACS hstcal applications.

closed 2017-10-04 by jamienoss