spacetelescope/acstools #42: Pass subprocess.call() results to user

jamienoss, 2017-11-02:
In acstools, all of the ``subprocess.call()`` that execute the underlying HSTCAL C executables, never return stdout nor stderr. When the called exe fails it does so silently.

For example, [ascrej()](https://github.com/spacetelescope/acstools/blob/master/acstools/acsrej.py#L187)
```Python
...
  if newbias:
        call_list.append('-newbias')

    if exe_args:
        call_list.extend(exe_args)

    subprocess.call(call_list)

def getHelpAsString():
    """
...
```

Perhaps we should bump this up to using ``subprocess.run()``, e.g.
```python
self.results = subprocess.run(cmd, shell=True, check=False, universal_newlines=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
```

__or__ ``subprocess.check_output()``

__or__ both e.g. something like this

```pyhton
try:
        results = subprocess.run(cmd, shell=True, check=False, universal_newlines=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
        if not results.returncode:
            return results.stdout.rstrip()
    except AttributeError:
        try:
            results = subprocess.check_output(cmd, shell=True, universal_newlines=True, stderr=subprocess.PIPE)
            return str(results).rstrip()
        except subprocess.CalledProcessError:
            return None
    except:
        return None
```

jamienoss, 2017-11-02:
@mdlpstsci FYI

pllim, 2017-11-02:
`shell=True` is discourage, I recommend seeking alternative from those available at https://docs.python.org/3/library/subprocess.html if you want to capture output.

PR #48 (2017-11-27): Raise an exception for non-zero returns from subprocess.call
Fixes #42
commit e38411bc (2017-11-27): Raise an exception for non-zero returns from subprocess.call

Fixes #42
closed 2017-11-27 by jamienoss