spacetelescope/acstools #47: acs_destripe_plus extract wrong dimension

pllim, 2017-11-20:
As reported by Norman:

*might have sniffed out a bug in acs_destripe_plus... tried it on a new-epoch subarray,  **jdd701a3q_flt.fits**... it seems to be doing bad math in `_get_ref_section`, returning something with dimension 2047 instead of 2048.*

pllim, 2017-11-20:
Update: Unable to find `_get_ref_section` function in the code base. Asked for clarification.

nagrogin, 2017-11-20:
Hi again,

I, this was probably from referring to my pre-official version of the code.  That identifier probably no longer exists, in the release version.

However, the error looked to be the same: the subarray dimension seem to be computed at 2047 when 2048 was needed.

Are you able to reproduce the error, or does your release version work correctly with that file I specified?

Thanks,

--n

pllim, 2017-11-20:
@nagrogin , thank you for the info. I am still waiting for MAST to deliver the file before I can check. 😅

nagrogin, 2017-11-20:
If this is dragging out, I can point you to the direct path once I get back to the office.

--n

pllim, 2017-11-21:
Okay, I am able to reproduce the error; Will investigate:
```
Traceback (most recent call last):
  File ".../bin/acs_destripe_plus", line 11, in <module>
    load_entry_point('acstools==2.0.9.dev5', 'console_scripts', 'acs_destripe_plus')()
  File ".../acstools/acs_destripe_plus.py", line 723, in main
    clobber=options.clobber, verbose=not options.quiet)
  File ".../acstools/acs_destripe_plus.py", line 499, in destripe_plus
    rpt_clean=rpt_clean, atol=atol, clobber=clobber, verbose=verbose)
  File ".../acstools/acs_destripe.py", line 468, in clean
    clobber=clobber, verbose=verbose)
  File ".../acstools/acs_destripe.py", line 513, in perform_correction
    frame = StripeArray(image)
  File ".../acstools/acs_destripe.py", line 100, in __init__
    self.configure_arrays()
  File ".../acstools/acs_destripe.py", line 114, in configure_arrays
    self.ingest_dark()
  File ".../acstools/acs_destripe.py", line 152, in ingest_dark
    self.dark = extract_dark(self.hdulist[0].header, self.hdulist[1])
  File ".../acstools/utils_calib.py", line 60, in extract_dark
    dark = extract_ref(scihdu, hdudark['sci', 2])
  File ".../acstools/utils_calib.py", line 240, in extract_ref
    'is {1}'.format(refdata.shape, (ny, nx)))
ValueError: Extracted reference image is (1024, 2048) but science image is (1024, 2072)
```

nagrogin, 2017-11-21:
Hi again,

P. L. Lim wrote:
> |File ".../acstools/utils_calib.py", line 240, in extract_ref 'is 
> {1}'.format(refdata.shape, (ny, nx))) ValueError: Extracted reference 
> image is (1024, 2048) but science image is (1024, 2072) |

This is not *exactly* reproducing the error I got (reprinted below), but 
similar:

>   File 
> "/usr/stsci/ssbdev/python/lib/python2.7/site-packages/acstools-2.0.0-py2.7.egg/acstools/acs_destripe.py", 
> line 287, in ingest_dark
>     self.science = self.science - self.dark * self.darktime
> ValueError: operands could not be broadcast together with shapes 
> (1024,2048) (1024,2047)

Cheers,

--n

pllim, 2017-11-21:
I suspect it is caused by the same bug, but at a different place now since we are using different versions and there had been some refactoring in between. My traceback has `acstools-2.0.9.dev5-py3.6.egg` (sorry, I scrubbed that off the text above to keep line length to minimum) while yours is `acstools-2.0.0-py2.7.egg`.

pllim, 2017-11-21:
Turns out this is not a bug in ACSTOOLS but rather the RAW file has outdated OSCNTAB. When I update it to point to the latest, nothing failed:
```python
with fits.open('jdd701a3q_raw.fits', mode='update') as pf:
   ...:     pf[0].header['OSCNTAB'] = 'jref$17717071j_osc.fits'
```

closed 2017-11-21 by pllim

nagrogin, 2017-11-21:
phew!
I wonder when the OSCNTAB would be updated for the /ifs directories...

thanks for clearing this up!

--n

pllim, 2017-11-21:
FWIW, the FLT I requested from MAST did have the correct and up-to-date OSCNTAB.

nagrogin, 2017-11-22:
P. L. Lim wrote:
>
> FWIW, the FLT I requested from MAST did have the correct and 
> up-to-date OSCNTAB.
>

Argh, my pull of the RAWs from MAST returns exactly the same
files as in /ifs --- with the obsolete OSCNTAB.  :(

I'm having a hell of a time, trying to properly rejigger the headers
so that acs_destripe_plus will de-stripe and CTE-correct these
modern subarray images.  I'm currently stuck on what DRKCFILEs
I'm supposed to be using (the images use various DARKFILEs).

Not able to find a suitable worked-example anywhere.  :(

Giving up for today,

--n