spacetelescope/acstools #54: calacs in notebooks messes up logging

tddesjardins, 2018-04-19:
When running `calacs` as part of a jupyter notebook, subsequent tasks that use the `logging` module (e.g., `astrodrizzle`) start printing all logging messages to the STDOUT as well as to the logfile they create in addition to the normal messages they print to STDOUT. This has the effect of duplicating much of the information these programs normally print. I've attached an example screenshot. 

[image]

Through testing of individual cells in the notebook, I've narrowed down that it's the cell that imports and runs `calacs` that is causing this behavior. The cell contains only:

```
from acstools import calacs

calacs.calacs('j9l960010_asn.fits')
```

If I save the FLT files `calacs` produces, restart the kernel in the notebook and clear all outputs, and skip the cell that imports and runs `calacs`, then I don't get this behavior with the `logging` messages and only things that are normally printed to STDOUT appear in the output cell.

jamienoss, 2018-04-19:
* Has this only _just_ started happening?

jamienoss, 2018-04-19:
I also don't understand the interplay, here, between the diff cells.

Your screenshot above is from tweakreg and shows some degree of duplicated messaging, but if calacs is being executed from a diff cell then how does it have anything to do with the tweakreg cell and its duplicated messages?

pllim, 2018-04-19:
As far as I can see, `calacs.py` wrapper simply calls `subprocess.check_call()`. @tddesjardins , does this behavior persist if you run `calacs.e` in your notebook *without* the wrapper, for example:
```
!calacs.e blahblahblah
```
?

pllim, 2018-04-19:
A workaround for now is to manually set logging level in `drizzlepac` to WARNING or ERROR (see https://docs.python.org/3/library/logging.html#levels). @mcara , how does one control logging behavior in `drizzlepac`?

mcara, 2018-04-19:
@pllim From what I see one cannot control anything about logging in astrodrizzle (well, except for file name). Level of warning is hardcoded like `log.info()`, `log.warn()`, etc.

pllim, 2018-04-19:
@mcara, is there no way do use `log.setLevel()`?

mcara, 2018-04-19:
Maybe you can try doing something like:
```
from drizzlepac import astrodrizzle
astrodrizzle.log.setLevel(logging.ERROR)
```

mcara, 2018-04-19:
@pllim I tried this and it seems to have no effect. Maybe @stsci-hack knows the secret sauce.

pllim, 2018-04-19:
@mcara thanks for checking. Maybe I should open an issue over at `drizzlepac` about the logging level aspect.

tddesjardins, 2018-04-19:
From discussions with @pllim and @jamienoss , my thinking now is that this behavior is due to the interaction between `subprocess.check_call()` and `jupyter`. I think this because the output from the `acstools` wrapper is sent to the terminal running the `jupyter` server, while directly calling the C executable inside the notebook directs STDOUT to the notebook's output cell. I'll need to test it more to confirm if that's the case, but in the meantime @pllim 's suggestion to directly call on the C executable works.

pllim, 2020-03-27:
@tddesjardins , did you find out anything more about this? Any action is needed on the `acstools` side? If not, please close the issue.

closed 2020-03-31 by pllim