spacetelescope/acstools #55: Bitwise error with satdet.update_dq().

KerryPaterson, 2018-05-08:
I get the error **TypeError: ufunc 'bitwise_and' not supported for the input types, and the inputs could not be safely coerced to any supported types according to the casting rule ''safe''** from line 878 (old mask = (dqval & dqarr) != 0) when trying to run satdet.update_dq().

pllim, 2018-05-09:
Hello. Are you trying to analyze HST data or something else? This package is primarily used to support HST ACS instrument.

Can you please provide a code snippet to minimally reproduce your problem? If you do not wish to disclose your data here on GitHub, please send them to `help[at]stsci.edu` email address. Thanks!

pllim, 2018-08-07:
Hello. I haven't heard back from you, so I am closing this. Feel free to re-open if you still have issues. Just so we're clear, this package is not guaranteed to work if you are not using it to process HST data, specifically those taken with ACS/WFC.

closed 2018-08-07 by pllim