spacetelescope/acstools #66: no option in acs_destripe_plus to retain BLV_TMP/BLC_TMP files

jryon, 2018-08-23:
`acs_destripe_plus` does not have an option to retain BLV_TMP/BLC_TMP files. Such a feature would be useful for users running the pixel-based CTE forward model on subarray images, since that step should be run after bias correction but before dark correction and flat-fielding.

pllim, 2018-08-23:
Just to be clear, do you want to retain the `bl?_tmp` files that are before or after destriping or both?

For the record, `calacs.e` has a command line option to keep temporary files. Unfortunately, this option does not exist for the individual `acsccd.e` and so on. Thus, this feature has to been implemented on the Python side of things.

Do you have any preference for the new keyword option? `keep_tmp`, `keep_temporary_files`, `keep_blctmp_and_blvtmp`, ...? (How much do you like to type? :wink: )

jryon, 2018-08-23:
I want the `bl?_tmp` files that have been destriped.

Probably shorter is better for the keyword option, so `keep_tmp` works for me.

pllim, 2018-08-24:
I am waiting to hear back about #46 before working on this one to avoid conflicts.

pllim, 2018-08-28:
@nmiles2718 , FWIW, if you want to use this new option via TEAL, it is better off as a separate keyword in the Python API. But we can table this discussions until a fix is available upstream. Thanks for looking into this!

pllim, 2020-03-27:
Revisiting this... I think our conclusion from #46 is that there is no flag we can pass directly to `acs2d` to keep the temporary files. If you look at the C program, the option to keep temp files is controlled by `calacs.e`, not `acs2d.e`. So, there are 3 options:

1. If you don't need this feature anymore, we will just close this as won't fix.
2. We can ask HSTCAL maintainer to change the code in ACS2D to support this feature and then `acstools` simply pass the new flag to `acs2d.e`.
3. We implement an ugly hack on the Python side here to make a copy of the file so it won't get deleted.

Please let me know what you think.

jryon, 2020-03-31:
I'd prefer to fix this issue than close it, so I'm fine with either 2 or 3.

nmiles2718, 2020-03-31:
I also prefer to fix it and I think the simplest route is via option 3.

nmiles2718, 2020-04-01:
I also perfer option 2 and I opened an issue in hstcal (https://github.com/spacetelescope/hstcal/issues/346) to address this issue across all our underlying C-executables. But realistically, it will take much longer to implement and this update is likely only to be used by the ACS Team, since most users only care about the final products (FLT/FLC, CRJ/CRC).

pllim, 2020-04-01:
Re: options -- This is an actively evolving situation, please see #120

PR #120 (2020-04-08): ENH: Enable CRREJ on acs_destripe_plus products, keep tmp files
This is basically #46 without the TEAL stuff.

Fix #35 
Fix #66

We thought `acs2d` was removing input files on the HSTCAL side but I don't think that is the case. I actually don't see the disappearing `blv_tmp` reported by @nmiles2718 in https://github.com/spacetelescope/acstools/pull/46#issuecomment-416599585 when I run this myself on a RAW file I have at hand.

**TODO**

- [ ] Get approval from ACS Team after they test this PR out. cc @jryon
closed 2020-04-08 by pllim