spacetelescope/acstools #68: Update documentation

nmiles2718, 2018-08-30:
Currently, the documentation for acstools contains no references to SINKCORR in the [calacs.e](https://acstools.readthedocs.io/en/stable/calacs_hstcal.html) section. In that same section, it also tells users that CALACS may be downloaded as part of the HSTCAL package in STSDAS. So we should update the docs to include links to astroconda and the SINKCORR step.

pllim, 2018-08-30:
xref spacetelescope/hstcal#318

cc @stscicrawford and @SaOgaz

nmiles2718, 2018-11-14:
In addition to adding documentation for SINKCORR, documentation should also be added here for the forward modeler. @jryon has developed a notebook that is a step-by-step guide to running the forward modeler, but we need additional docs setup in this repo so that the new interface shows up on the [acstools website](https://acstools.readthedocs.io/en/stable/).

jryon, 2018-11-14:
Who's responsibility is it to update acstools documentation?

pllim, 2018-11-14:
It is open source. Feel free to submit PRs!

PR #79 (2019-02-07): SINKCORR and forward model doc updates
Minor updates to the documentation. Includes documentation for SINKCORR and ACSCTEFORWARDMODEL. Closes #68
closed 2019-02-07 by pllim