spacetelescope/acstools #69: TST: Move tests off pandokia

pllim, 2018-09-14:
Tests to port: https://ssb.stsci.edu/pandokia3/pandokia.cgi?query=treewalk&attn=%2A&compare=0&context=dev%3A3.5.6&host=nott&project=stsci_python&status=%2A&test_name=acstools%2F%2A&test_run=conda-conda_tests_py3_2018-09-14-10%3A30%3A02

For the HSTCAL wrappers, also see spacetelescope/hstcal#341

pllim, 2018-09-17:
I'll let you lead this one, @mdlpstsci , but will be glad to assist where needed.

If you want to test wrapper here, we have 2 options:

1. Modify Jenkins file(s) to install HSTCAL from source but only CALACS (there is a configuration option for that in `waf`).
    * Do we also need to test against released HSTCAL in AstroConda?
2. Mock HSTCAL executables with [mock](https://docs.python.org/3/library/unittest.mock.html). This does not require HSTCAL to be installed and would test the Python portion only.

I have a feeling that @stscicrawford meant Option 1.

mdlpstsci, 2018-09-19:
@pllim I have a wrapper which I would like to test so I now have to understand how to implement Option 1.  I have assumed that the implementation is similar to what is done for the HSTCAL Jenkinsfile, but I need some advice/assistance (Thursday? via phone).

pllim, 2018-09-19:
@mdlpstsci , sure. Happy to help!

pllim, 2018-09-20:
@rendinam wrote up a guide at https://innerspace.stsci.edu/display/SSR/CI+System+Usage+Notes#CISystemUsageNotes-BuildConfiguration . Hope this helps!

pllim, 2018-09-20:
Also the guide's example might be dated by now. Feel free to open a PR and I can help Jenkinsfile get the Groove back if needed. :smile:

mdlpstsci, 2018-09-20:
Thanks.  Will read the background material.  You will eventually have the joy of my Jenkinsfile.

closed 2020-03-27 by pllim