spacetelescope/acstools #77: Enhancement: Zeropoint Calculator API

nmiles2718, 2019-01-02:
The [ACS Zeropoints Calculator]( https://acszeropoints.stsci.edu/) is a webtool developed by the ACS Team to provide users with an interface to compute time-dependent zeropoints without having to download the required component files from CRDS. Users submit a `date` and `detector` which together comprise an "obsmode" used by `pysynphot` to compute the `STmag`, `VEGAmag`, `ABmag` zeropoints for all the ACS filters (or just a single one).

We have had several users ask (e.g. INC0130965) if there is anyway to script the process of computing zeropoints for a large number of date and detector combinations. I have re-written the Zeropoints Calculator to work using `GET` methods, rather than `POST`. Hence, the user supplied information is now passed via a query string embedded in the URL which means we can finally allow users to batch process things. Next, to give users tools to utilize these updates I wrote a new module in `acstools` called `acszpts` that allows users to programmatically query the Zeropoints Calculator and returns the results as an `astropy.table.Table` with units. The exact API is still a work in progress and I am open to any suggestions. I am opening this issue to keep track of things and provide a forum for discussion.

Example Usage:
```python
In [1]: from acstools import acszpt
The following tasks in the acstools package can be run with TEAL:
          acs2d                  acs_destripe           acs_destripe_plus     
          acsccd                    acscte              acscteforwardmodel    
          acsrej                    acssum                    calacs
PixCteCorr is no longer supported. Please use acscte.

In [2]: zpt_query = acszpt.Query(date='2016-04-04',detector='WFC')

In [3]: zpt_query.submit_request()

In [4]: zpt_query.parse_response()

In [5]: zpt_table = zpt_query.format_results()

In [6]: zpt_table
Out[6]: 
<Table length=13>
FILTER PHOTPLAM        PHOTFLAM         STmag  VEGAmag  ABmag 
       Angstrom erg / (Angstrom cm2 s) mag(ST)   mag   mag(AB)
 str6  float64         float64         float64 float64 float64
------ -------- ---------------------- ------- ------- -------
 F435W   4329.2              3.148e-19  25.155  25.763  25.665
 F475W   4746.2              1.827e-19  25.746  26.149  26.056
 F502N   5023.0              5.259e-18  22.098  22.365  22.285
 F550M   5581.5               3.99e-19  24.898  24.825  24.856
 F555W   5360.9              1.963e-19  25.667  25.713  25.713
 F606W   5922.0              7.811e-20  26.668  26.405  26.498
 F625W   6312.0              1.188e-19  26.213  25.735  25.904
 F658N   6584.0               1.97e-18  23.164  22.381  22.763
 F660N   6599.4              5.156e-18  22.119  21.428  21.714
 F775W   7693.2              9.954e-20  26.405  25.272  25.667
 F814W   8045.0              7.046e-20   26.78  25.517  25.944
F850LP   9033.2               1.52e-19  25.945  24.332  24.858
 F892N   8914.8              1.502e-18  23.458  21.905    22.4
```

tddesjardins, 2019-01-02:
Following up on the conversation I had with @nmiles2718 earlier, it would be great if we add a table of the aperture correction values and the photometric CTE calculator as well. This could be a module in `acstools` called something like `phot_tools` with sub-modules for each of the zeropoint calculator, CTE calculator, and the aperture correction tables.

pllim, 2019-01-04:
Pull request is very welcome! 😉

nmiles2718, 2019-01-04:
It's coming! Just waiting on the updates I made to the calculator to be pushed into production. Once that is done, I'm going to finish up the documentation and it should be good to go. Oh, and by it, I mean the `acszpt` module. We can worry about adding the extra features later as time permits. 😅

PR #78 (2019-02-01): Zeropoint API
I have added a new module for `acstools` called `acszpt`, which address the need for an interface to programmatically query the [ACS Zeropoints Calculator](https://acszeropoints.stsci.edu/). The module is designed to only work with valid combinations of the ACS instruments, their filters, and observation dates. If the users query fails to meet the proper criteria, an error is raised along with messages for each input that failed.  

closes #77
closed 2019-02-01 by pllim