spacetelescope/acstools #123: interpret_bit_flags should be imported from astropy.nddata.bitmask

jdavies-st, 2020-04-27:
The imports

https://github.com/spacetelescope/acstools/blob/174a5953de431aadcaab8c1014610806ddad8554/acstools/acs_destripe_plus.py#L276-L279

https://github.com/spacetelescope/acstools/blob/b5e240e1703f92a73bd482d08b3f670deeb03f7f/acstools/acs_destripe.py#L544-L551

and other `bitmask` imports should be migrated to 

```
from astropy.nddata.bitmask import interpret_bit_flags
```

and the like.  This module has been migrated to `astropy` since 3.1.

pllim, 2020-04-27:
@mcara , you wrote this code, right?

pllim, 2020-04-27:
Either we need to bump

https://github.com/spacetelescope/acstools/blob/bc40b09963cb2301eb314494d3ed62695b26a3b6/setup.cfg#L31

or there needs to be a compat layer. But regardless, this won't remove the dependency of `stsci.tools` because `acstools` still uses other parts of it.

mcara, 2020-04-27:
Yes, I wrote it. What happened? `stsci.tools`? I would say it is safe to set `astropy>=2` and just rely on `astropy.nddata.bitmask`

PR #124 (2020-04-28): MNT: Use bitmask from astropy>=3.1
Fix #123 

cc @mcara and @jdavies-st
closed 2020-04-28 by pllim