spacetelescope/acstools #125: Extensions of CRJ_TMP files must be divided by number of combined images in crrej_plus

jryon, 2020-05-06:
In `acs_destripe_plus.crrej_plus`, after `acsrej` is run on the input BLV_TMP files to produce the CRJ_TMP file, the SCI and ERR extensions of the CRJ_TMP file must be divided by the number of input BLV_TMP files. This is because `acsrej` basically outputs a sum of the input files (modulo CR rejections). When `acs2d` is run on this sum, an incorrect amount of dark and flash are removed.

Apologies for not noticing this bug earlier. It only came to my attention when checking flashed subarrays processed with `crrej_plus` because the flash wasn't completely removed.

pllim, 2020-05-06:
That's surprising. I'll have to dig at HSTCAL to see how it is doing this. Python wrapper shouldn't have to implement own code around the actual pipeline. Hmm. Thanks for reporting this!

jryon, 2020-05-06:
Yeah, I looked into HSTCAL to make sure this wasn't their issue, but I have trouble reading C ;) Also, I made sure that this behavior also affects full-frame data when running `acs2d` on results from `acsrej`, and it does.

pllim, 2020-05-06:
Does `ASN_MTYP` mean anything to you? If I remember correctly, HSTCAL works exclusively with ASN file for a group of images. And in this code below, it seems to modify `ASN_MTYP` before calling ACSREJ.

https://github.com/spacetelescope/hstcal/blob/11b1be97b6a6797e9146e087ef3dd2e7909ce64a/pkg/acs/calacs/calacs/calacs.c#L701-L709

```c
                /* Set up the correct value of ASN_MTYP for ACSREJ
                   output image. */
                if ( strncmp(acshdr->asn_table, acshdr->crj_tmp, 9) ) {
                    /* crj_tmp is a sub-product */
                    strcpy (acshdr->mtype, asn->product[prod].subprod[posid].mtype);
                } else {
                    /* crj_tmp leads to a product with same rootname as asn_table */
                    strcpy (acshdr->mtype, asn->product[prod].mtype);
                }
```

That said, does this bug exist if your input is ASN file? How are you running the functions exactly? Thanks!

jryon, 2020-05-06:
`ASN_MTYPE` doesn't mean anything to me. I was running `crrej_plus` with a list of BLC_TMPs. I haven't tried an ASN simply because it takes a decent amount of effort to make one :(

Can't we run `acstools.calacs` with a list of images? I guess I'd be surprised if CRJs processed from an input list have this problem, but maybe no one does this for flashed images?

pllim, 2020-05-06:
> Can't we run `acstools.calacs` with a list of images? 

I don't think so? The input is passed straight into `calacs.e`. If we could, we wouldn't have recommended a loop at https://acstools.readthedocs.io/en/latest/calacs.html#batch-calacs ?

pllim, 2020-05-06:
I have a few clarifying questions:

1. Does CALACS give the correct result when this group of inputs go through ACSREJ/ACS2D using the C pipeline? Like, you run `calacs.e` on them and the final CRJ/CRC is correct?
2. Does `crrej_plus` here give different results when you give it a list of BLV_TMP vs the ASN file?

I am worried that by implementing a workaround here, we might:

1. Mask the real underlying problem.
2. Introduce bug in other uses cases.

pllim, 2020-05-06:
p.s. Does `dodark.c` not take account of input EXPTIME anymore in CALACS? ACSREJ does adjust the EXPTIME value when it combines the inputs. I thought the dark current would be scaled accordingly before subtraction. My memory is rusty and the code has changed. I'll loop in @mdlpstsci for advice.

pllim, 2020-05-06:
[DHB](https://hst-docs.stsci.edu/acsdhb/chapter-3-acs-calibration-pipeline/3-4-calacs-processing-steps#id-3.4calacsProcessingSteps-3.4.43.4.4acs2d) does mention scaling by EXPTIME for `doDark` and only mentions FLASHDUR for `doFlash`. I am not sure how the latter plays with ACSREJ.

jryon, 2020-05-06:
Ah, ok, this is clarifying things.

1. I don't think you can run `calacs.e` on a list of images (that will become a CRJ). At least, I don't know how to. Maybe that explains why ASNs exist.
2. I will test this.

Yes, EXPTIME is definitely taken into account for dark correction. FLASHDUR is not adjusted by ACSREJ, according to my testing, so flash correction of an improperly-normalized CRJ will be wrong. There could also be some weirdness with flat correction of an improperly-normalized CRJ.

pllim, 2020-05-06:
If the bug turns out to be intrinsic to the C pipeline, it should be fixed over at HSTCAL, not here. Thanks for checking! I'll wait to hear back.

pllim, 2020-05-06:
Am I going crazy or scaling by EXPTIME was completely removed in spacetelescope/hstcal#467 ?

mdlpstsci, 2020-05-06:
HI,
You are correct Pey-Lian.  The exposure time is no longer used for dodark PR#467.    The darktime value is nominally the darktime keyword from the header which can be modified by a darktime_offset for the CCDs only.
Michele

Michele D. De La Pena
Space Telescope Science Institute
mdelapena@stsci.edu<mailto:mdelapena@stsci.edu>
Mobile: 520.419.8825
Office: 667.218.6501

pllim, 2020-05-07:
After extensive discussions with the ACS Team, it seems that this is a bug upstream at CALACS (HSTCAL). There are actually two problems:

1. DARKTIME not properly propagated in ACSREJ before being passed into DARKCORR in ACS2D (introduced in spacetelescope/hstcal#467).
2. FLASHDUR not properly propagated in ACSREJ before being passed into FLSHCORR in ACS2D (a long pre-existing condition).

pllim, 2021-02-26:
Just to follow up: Has this been fixed yet in HSTCAL?

jryon, 2021-02-26:
No, not yet.

pllim, 2021-06-22:
This issue is now a year old. Just want to check again that it is still relevant here.

jryon, 2021-06-28:
The underlying issues related to DARKTIME and FLASHDUR for CRJs are being handled in CALACS: https://github.com/spacetelescope/hstcal/pull/551
https://github.com/spacetelescope/hstcal/pull/550

I think this issue can be closed.