spacetelescope/acstools #137: Arbitrary upper limit on input date in acszpt

jryon, 2020-10-15:
The `acszpt` module requires that the input date of the calculation be earlier than 12-31-2021: https://github.com/spacetelescope/acstools/blob/62c02b3df3534a89a070d450d6759bc187cdd07c/acstools/acszpt.py#L168-L170

The ACS zeropoints tool webapp does not throw an error or warning for the middle of Cycle 29 (04-01-2022): https://acszeropoints.stsci.edu/results_all/?date=2022-04-01&detector=WFC

There is a date upper limit built into the throughput tables, so perhaps the `acszpt` module or the zeropoints tool could read that date from those files?

jryon, 2020-10-15:
We probably don't need to bug @nmiles2718. Maybe arbitrary was the wrong word. The date is correct for old versions of the throughput files, but they have since been updated to extend to ~2030.

pllim, 2020-10-15:
Yes, please open a pull request. Thanks!

molaes-stsci, 2020-10-15:
> We probably don't need to bug @nmiles2718. Maybe arbitrary was the wrong word. The date is correct for old versions of the throughput files, but they have since been updated to extend to ~2030. I'll take a look at how to fix this issue soonish, and maybe I can get some help from @molaes-stsci?

@rjavila brought this up with me the other day in terms of the webtool on gitlab. I vaguely remember @nmiles2718 bringing this up, it will take me a moment to familiarize myself but should be a simple fix.

PR #140 (2020-11-10): Updated handling of extrapolated dates
- fixed minor typo in example
- removed usage of locally defined "extrapolation_date"
- added functionality to pass warnings that have been generated from ZPT tool, if any
Fixes #137
closed 2020-11-10 by pllim