spacetelescope/acstools #162: api supporting acszpt seems to have changed.

jackwarfield, 2022-08-08:
### Description
The acszpt module no longer works because of changes in the api at acszeropoints.stsci.edu

### Expected behavior
Return a table with zeropoint information.

### Actual behavior
Error, as the page being queried is no longer accessible,

### Steps to Reproduce
You can reproduce the error by running the python example code:

```python
from acstools import acszpt
date = '2016-04-01'
detector = 'WFC'
filt = 'F435W'
q = acszpt.Query(date=date, detector=detector, filt=filt)
zpt_table = q.fetch()
```
Which returns

```
INFO:acszpt.Query:Checking inputs...
INFO:acszpt.Query:Submitting request to https://acszeropoints.stsci.edu/results_single/?date1=2016-04-01&detector=WFC&WFC_filter=F435W
ERROR:acszpt.Query:<HTTPError 403: 'Forbidden'>
-------------------------------------------------------------------------------
The query failed! Please check your inputs. If the error persists, submit a ticket to the ACS Help Desk at hsthelp.stsci.edu with the error message displayed above.
```

### System Details

macOS-12.4-x86_64-i386-64bit
Python 3.9.2 | packaged by conda-forge | (default, Feb 21 2021, 05:02:20)
[Clang 11.0.1 ]
Numpy 1.21.5
astropy 5.0.2
Scipy 1.8.0
Matplotlib 3.5.1
stsci.tools 4.0.1
ACS Tools 3.3.1

pllim, 2022-08-08:
Thank you for reporting this! I have informed the ACS Team and they are looking into a fix.

closed 2022-08-08 by jryon

pllim, 2022-08-08:
@jackwarfield , this should now be fixed in the dev version. We're working on a new release soon. Thanks!

pllim, 2022-08-09:
@jackwarfield , we have just released 3.4.0 with the fix. Does it work for you now? `pip install acstools -U`

https://github.com/spacetelescope/acstools/releases/tag/3.4.0

jackwarfield, 2022-08-09:
@pllim It seems to be working fine and normal now. Thanks for the quick work on this!