spacetelescope/acstools #166: Release: Update Zenodo DOI on README after the next release

pllim, 2022-08-09:
I have just activated Zenodo webhook for this repo. This means that GitHub Release is now going to automatically push the tarball to Zenodo for acstools. This also means that when that happens in the *next* release (the one after 3.4.0), you will get a new Zenodo DOI. You need to update the README for this repo to point to that new DOI, which will not be generated until after the *next* release.

cc @jryon @rjavila

pllim, 2022-12-07:
Well, on the right side, there is still small print if you are interested:

Cite all versions? You can cite all versions by using the DOI [10.5281/zenodo.7406932](https://doi.org/10.5281/zenodo.7406932). This DOI represents all versions, and will always resolve to the latest one. =

jryon, 2022-12-07:
That right side text is also present on the v3.4.0 page, but has a different DOI and resolves to v3.4.0: https://zenodo.org/record/6977374

pllim, 2022-12-07:
Yeah... the DOI page generated by auto release knows nothing about your past manual releases. If that bothers you and you want them merged, you have to contact Zenodo help desk.

jryon, 2022-12-07:
Ok, that's the bit that wasn't clear to me. Thanks for clarifying. I don't think we mind, just wanted to be sure things are as expected.

jryon, 2023-05-08:
I think this is closed by #173

closed 2023-05-08 by jryon